Groupsun/CCPA-2020-LAB #2: 关于Verilog⽂件生成位置的提问

ForTheFreedom, 2020-05-14:
请问Emitter.dumpVerilog(Emitter.dump(Emitter.emit(HalfAdder()),
"HalfAdder.fir"))语句生成的Verilog⽂件在哪

Groupsun, 2020-05-14:
默认在同级文件夹中生成一个.fir文件夹，这个文件夹默认是隐藏的，你可以在终端里面cd进去，也可以在文件管理器使用ctrl+alt+H查看隐藏的文件夹和文件。

Groupsun, 2020-05-14:
实际上，我在设计这个接口的时候有点疏忽部分，我将生成的.fir以及verilog文件都默认放进了写死的同级.fir文件夹当中去了。但是考虑到`dumpVerilog()`以及`dump()`两个接口是分离的，这种方法会方便我实现，今后的扩展也要方便一些。
你可以详细参见pyhcl源代码当中：https://github.com/scutdig/PyChip-py-hcl/blob/master/pyhcl/dsl/emitter.py

```python
# /dsl/emitter.py
# ...
@staticmethod
def dump(s, filename) -> str:
    if not os.path.exists('.fir'):
        os.mkdir('.fir')

        f = os.path.join('.fir', filename)
        with open(f, "w+") as fir_file:
            fir_file.write(s)

            return f

        @staticmethod
        def dumpVerilog(filename):
            os.system('firrtl -i %s -o %s -X verilog' % (filename, filename))
# ...
```

这一部分的内容。

closed 2020-09-16 by Groupsun