Groupsun/CCPA-2020-LAB #5: sbt compile指令长时间没回应

RedLotus-jpg, 2020-05-14:
[image]
这个好像停在这很久了，怎么处理，有没有影响的。
[image]
另外sbt test指令中途会kill，跳了一连串的warn出来，同问。

Groupsun, 2020-05-15:
sbt compile长时间没有回应，上面是由于配置sbt下载速度太慢了，需要你更换sbt源来处理，注意不是apt源是sbt源，教程可以参照：https://www.jianshu.com/p/8b4609d57f0a
后者中途会kill掉，是因为你虚拟机划分的内存太小了，建议将内存划到4G来执行。

RedLotus-jpg, 2020-05-15:
> sbt编译没有持续响应，上面是由于配置sbt下载速度太慢了，需要你更换sbt源来处理，注意不是aptsource是sbt源，教程可以参照：[https](https://www.jianshu.com/p/8b4609d57f0a) ://www.jianshu.com/p/ [8b4609d57f0a](https://www.jianshu.com/p/8b4609d57f0a)
> 附件中途会kill掉，是因为你虚拟机划分的内存太小了，建议将内存划到4G来执行。

[image]
这个应该算是装好了吧，但却找不到sbt目录。

RedLotus-jpg, 2020-05-15:
> sbt编译没有持续响应，上面是由于配置sbt下载速度太慢了，需要你更换sbt源来处理，注意不是aptsource是sbt源，教程可以参照：[https](https://www.jianshu.com/p/8b4609d57f0a) ://www.jianshu.com/p/ [8b4609d57f0a](https://www.jianshu.com/p/8b4609d57f0a)
> 附件中途会kill掉，是因为你虚拟机划分的内存太小了，建议将内存划到4G来执行。

[image]
这个样子应该算是装好了吧，但是我无法找到sbt目录。

Groupsun, 2020-05-15:
> > sbt编译没有持续响应，上面是由于配置sbt下载速度太慢了，需要你更换sbt源来处理，注意不是aptsource是sbt源，教程可以参照：[https](https://www.jianshu.com/p/8b4609d57f0a) ://www.jianshu.com/p/ [8b4609d57f0a](https://www.jianshu.com/p/8b4609d57f0a)
> > 附件中途会kill掉，是因为你虚拟机划分的内存太小了，建议将内存划到4G来执行。
> 
> [image]
> 这个样子应该算是装好了吧，但是我无法找到sbt目录。

只要你执行`sbt assembly`成功就行了，你进入/utils/bin，执行`./firrtl --help`试试看有没有出错，如果问题的话就可以了。接着把`<你所放在的路径>/firrtl/utils/bin`加入到环境变量当中即可。

RedLotus-jpg, 2020-05-15:
> 只要你执行`sbt assembly`成功就行了，你进入/utils/bin，执行`./firrtl --help`试试看有没有出错，如果问题的话就可以了。接着把`<你所放在的路径>/firrtl/utils/bin`加入到环境变量当中即可。

[image]
[image]
这样就算是没问题了吗

Groupsun, 2020-05-15:
> 
> [image]
> [image]
> 这样就算是没问题了吗

是的，接下来添加进环境变量当中即可。

lisanzhu, 2020-05-16:
> > [image]
> > [image]
> > 这样就算是没问题了吗
> 
> 是的，接下来添加进环境变量当中即可。

你好，请问一下我怎么将他添加进环境变量的时候不好用了，可是我已经将其加入其中了

Groupsun, 2020-05-16:
> 你好，请问一下我怎么将他添加进环境变量的时候不好用了，可是我已经将其加入其中了

同学能不能更详细的描述你的问题，提供一些结果图片之类的。

lisanzhu, 2020-05-16:
[image]
[image]
请您看一下谢谢啦

Groupsun, 2020-05-16:
你描述的问题于本issue问题无关，我帮你独立开一个新的issue来参考：https://github.com/Groupsun/CCPA-2020-LAB/issues/9

closed 2020-09-16 by Groupsun