Groupsun/CCPA-2020-LAB #7: Command 'sbt' not found

Faye1999, 2020-05-16:
已经安装了sbt
[image]

然后配置到环境中
[image]

但是执行sbt compile的时候出现 Command 'sbt' not found 
请问是什么问题呢？
[image]

Groupsun, 2020-05-16:
你在添加环境变量之后，是否有执行`source`命令？你可以使用`echo $PATH`来查看环境变量是否生效了。如果都不是上述这些问题的话，可以考虑重新启动终端、重新登录用户以及重启的方式解决。

Faye1999, 2020-05-16:
执行过source命令，环境变量应该没问题
[image]

执行sbt的时候，会出现Permission denied
[image]

试过这样添加权限，还是不行
[image]

不知道是不是哪里的权限出了问题？

Faye1999, 2020-05-16:
已经解决了...
把sbt解压到用户目录下就没问题

还是感谢助教！！

Groupsun, 2020-05-16:
你可以尝试用sudo执行sbt，如果你把sbt放在了系统目录下；不过你已经ok了就没什么问题。
还有些同学会遇到执行sbt命令权限不足的原因，有些是因为你的firrtl目录放在了系统目录下，这里建议firrtl目录统一放在用户目录下。

closed 2020-09-16 by Groupsun