Groupsun/CCPA-2020-LAB #8: clone  firrtl时遇到的问题

hzx718, 2020-05-16:
按照GitHub教程上安装firrtl，运行命令git clone https://github.com/freechipsproject/firrtl.git && cd firrtl出现以下问题，请问该如何解决
[image]

Groupsun, 2020-05-16:
你可以尝试执行下面的命令后再克隆：
```shell
git config --global http.postBuffer 524288000
```
如果问题依旧的话，可以考虑改成用ssh的方式来克隆。
这个问题可能是因为firrtl比较大，curl的postBuffer默认值太小的原因。
你还可以参照：
https://blog.csdn.net/IT_liuchengli/article/details/77040806

hzx718, 2020-05-16:
> 你可以尝试执行以下的命令后再克隆：
> 
> ```shell
> git config --global http.postBuffer 524288000
> ```
> 
> 如果问题依旧的话，可以考虑改成用ssh的方式来克隆。
> 这个问题可能是因为firrtl比较大，curl的postBuffer值太小的原因。
> 你还可以参照：[https](https://blog.csdn.net/IT_liuchengli/article/details/77040806) : //blog.csdn.net/
> [IT_六成里/文章/细节/ 77040806](https://blog.csdn.net/IT_liuchengli/article/details/77040806)

已经解决了，谢谢助教

hzx718, 2020-05-16:
> git config --global http.postBuffer 524288000
> ```
>
> 如果问题依旧的话，可以考虑改成用ssh的方式来克隆。
> 这个问题可能是因为firrtl比较大，curl的postBuffer值太小的原因。
> 你还可以参照：[https](https://blog.csdn.net/IT_liuchengli/article/details/77040806) : [...]

已经解决了，谢谢助教

Groupsun, 2020-05-16:
如果解决了先别close吧，我担心某些不会用的同学会不知道已经解决了的issue会被转移到closed里面，issue默认界面会过滤掉close了的。

hzx718, 2020-05-16:
不好意思，没看清楚，以为是关闭评论....

closed 2020-09-16 by Groupsun