Groupsun/CCPA-2020-LAB #9: firrtl环境变量配置问题

Groupsun, 2020-05-16:
original from: https://github.com/Groupsun/CCPA-2020-LAB/issues/5#issuecomment-629630642

> [image]
> [image]
> 请您看一下谢谢啦

Groupsun, 2020-05-16:
可以使用`echo $PATH`来验证环境变量有没有配置成功，而且推荐使用添加到`~/.bashrc`或者`/etc/profile`等方法来添加环境变量。
https://blog.csdn.net/White_Idiot/article/details/78253004

lisanzhu, 2020-05-16:
> 可以使用`echo $PATH`来验证环境变量有没有配置成功，而且推荐使用添加到`~/.bashrc`或者`/etc/profile`等方法来添加环境变量。
> https://blog.csdn.net/White_Idiot/article/details/78253004

好的谢谢您，我想要请问一下，在Mac中使用brew来安装sbt为什么下载好慢，一个多小时也是没有增长多少，请问这个是什么原因呢。
[image]

Groupsun, 2020-05-16:
> 好的谢谢您，我想要请问一下，在Mac中使用brew来安装sbt为什么下载好慢，一个多小时也是没有增长多少，请问这个是什么原因呢。
> [image]

brew在国内用由于不可抗力因素会很慢，可以换中科大的源来解决：https://lug.ustc.edu.cn/wiki/mirrors/help/brew.git

lisanzhu, 2020-05-16:
> > 好的谢谢您，我想要请问一下，在Mac中使用brew来安装sbt为什么下载好慢，一个多小时也是没有增长多少，请问这个是什么原因呢。
> > [image]
> 
> brew在国内用由于不可抗力因素会很慢，可以换中科大的源来解决：https://lug.ustc.edu.cn/wiki/mirrors/help/brew.git

好的，谢谢您

closed 2020-09-16 by Groupsun