Groupsun/CCPA-2020-LAB #11: 运行sbt assembly时遇到的问题

hzx718, 2020-05-17:
运行sbt assembly时一开始进度挺快，后面卡住不动了，请问是正常现象还是有什么问题？
[image]

[image]

Groupsun, 2020-05-17:
从截图来看单纯只是因为下载比较慢，但你之前说过已经更换了源了，所以我考虑一下会不会是你给虚拟机划分的内存太少了？（虽然这个可能性不大，因为上图看起来只是因为下载太慢了）
不过你卡住不动的话也确实有可能是因为内存不足造成的，只是还没有被kill掉。你试试划成4G的内存再重新执行一次试试，你还可以参考：https://github.com/Groupsun/CCPA-2020-LAB/issues/5#issue-618472910

hzx718, 2020-05-17:
更改内存后运行又遇到以下错误，这该怎么解决....
[image]
[image]

hzx718, 2020-05-17:
助教在吗

Groupsun, 2020-05-17:
从截图来看是平台不支持的问题，你提供一下你的java版本，完整的终端stdout输出（不要截图，把终端的文字全部复制出来，从你执行的命令开始）我再进一步看看怎么解决你的问题。我现在暂时查到的一个与你相关的问题在这：
https://github.com/os72/protoc-jar-maven-plugin/issues/68

这里提到的你可以先自己阅读一下。

hzx718, 2020-05-17:
huangzixi@huangzixi-VirtuaLBox:-$ java - version
openjdk version "11.0.7" 2020-04- 14
OpenJDK Runtime Environment (build 11.0. 7+10- post -Ubuntu- 3ubuntu1)
OpenJDK 64-Bit Server VM (build 11.0. 7+10- post- ubuntu- 3ubuntu1, mixed mode, sharing )

Groupsun, 2020-05-17:
这是我在Mac上的Java环境，我推测有可能是你使用的是openjdk的缘故。你可以试着更换成和我一样的版本来试试：

```
java version "11.0.5" 2019-10-15 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.5+10-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.5+10-LTS, mixed mode)
```

hzx718, 2020-05-17:
这要怎么更改，是换成oracle jdk吗

Groupsun, 2020-05-17:
是的，你可以先试试。

hzx718, 2020-05-17:
可是我问了一下同学，他的也是openjdk，为什么他的就没有这个问题呢

Groupsun, 2020-05-17:
因为这个问题本身我也不太清楚是怎么回事，我只能推测可能有这个原因在。因为openjdk是开源版本，它和oracle jdk是有差别的。我建议你针对这个问题自己去查找一下解决方法，如果实在不行的话还有最后一个方法，就是你让编译好的同学把他的整个firrtl文件夹发给你也行。

你还得留意一下你的openjdk版本和他的有没有区别，这个sbt是很古怪的，不同的jdk版本也有可能会导致不同的结果。

hzx718, 2020-05-17:
还是不行

Groupsun, 2020-05-17:
那你让你配置编译好的同学把整个firrtl文件夹发给你，然后你直接添加进环境变量里好了。

Groupsun, 2020-05-17:
我查看了一下firrtl那边的issues，那边推荐你的版本使用的是jdk 1.8，所以如果你愿意折腾的话就再试试看行不行。不愿意的话就按照我上面给你的方法去做。

hzx718, 2020-05-17:
好吧，谢谢助教

spana310, 2020-05-18:
我的Ubuntu是安装在移动硬盘上的。
在华为matebook e(arm处理器)上运行`sbt compile`时会出现下图中的错误，换到其他Intel处理器的电脑就success了，可能编译还是和处理器架构有关。
[image]

closed 2020-09-16 by Groupsun