Groupsun/CCPA-2020-LAB #13: 关于配置环境变量后，pycharm中不生效

lisanzhu, 2020-05-17:
在我配置结束firrtl环境变量之后，在我pycharm的编程中没有生效，出现了下列错误。而在我终端里面我的firrtl --help是可以得到结果的、我不清楚是不是我的项目结构出错了
[image]

Groupsun, 2020-05-17:
你应该是只配置了zsh的环境变量而没有配置shell的，你可以把pycharm的终端换成zsh的试试，或者你把firrtl也添加到shell的环境变量当中。

lisanzhu, 2020-05-17:
> 你应该是只配置了zsh的环境变量而没有配置shell的，你可以把pycharm的终端换成zsh的试试，或者你把firrtl也添加到shell的环境变量当中。

好的谢谢您。我刚才清了一下缓存之后好用了，程序也可以运行。不过我的代码还是有提示错误信息，请问这个正常嘛、
[image]

Groupsun, 2020-05-17:
这不是错误信息，这个已经是正常执行完了。输出的代码在`.fir`文件夹里面。

lisanzhu, 2020-05-17:
> 这不是错误信息，这个已经是正常执行完了。输出的代码在`.fir`文件夹里面。

好的，谢谢您

Onezity, 2020-05-22:
我也遇到了相同的问题，我在另一台重装时sbt publishlocal失败了，但是最终firrtl --help是有内容输出的，然后pycharm里就会显示not found（之前装的时候sbt publishlocal成功就不会有这样的问题）。然后我在.bashrc文件中是有添加环境变量的。然后我不大懂您上面说的配置shell具体是修改什么，在网上查的步骤感觉非常复杂，不大明白。

Groupsun, 2020-05-22:
你可以参照上面的同学清除缓存试试，或者在pycharm里面添加环境变量再试试。

ZoeDong, 2020-05-23:
请问添加环境变量是在Run->Edit Configurations->Environment variables把PATH添加进去吗？如果是的话尝试过清除缓存和添加环境变量，但是都没有效果。但是又确实是能够生成.fir文件所以不太明白这个报错到底有没有影响，运行后界面是这样的。
[image]

closed 2020-09-16 by Groupsun