Groupsun/CCPA-2020-LAB #18: 实验一

Grace-919, 2020-05-18:
[image]
 程序运行之后一直提示：
class HalfAdder (Module)
NameError: name "Module" is not defined

Groupsun, 2020-05-18:
你要把pyhcl源码包放在你项目的根目录下，并`from pyhcl import *`。

Grace-919, 2020-05-18:
> 项目的根目录下，并from pyhcl import *

[image]
这样放置pychl吗？但是还是有问题

Groupsun, 2020-05-18:
你需要把`pyhcl2`下面的`pyhcl`文件夹放在根目录下才行，这个才是源码包。

Grace-919, 2020-05-18:
> 你要把pyhcl源码包放在你项目的根目录下，并from pyhcl import *。
[image]
后来把pyhcl源码包放在项目的根目录（如上图）也不行。
最后重新建立.py文件，就可以了。
原来一直出错是因为开始的时候将文件建立在venv目录下了

closed 2020-09-16 by Groupsun