Groupsun/CCPA-2020-LAB #20: cocotb make中错误

sinceseto, 2020-05-18:
尝试用第0章pdf里的例子测试环境，请问这怎么解决？
makefile内容如下
```
VERILOG_SOURCES = $(PWD)/FullAdder.v
TOPLEVEL = FullAdder
MODULE = fulladder_test
SIM = verilator
EXTRA_ARGS += --trace

include $(shell cocotb-config --makefiles)/Makefile.inc
include $(shell cocotb-config --makefiles)/Makefile.sim
```
文件放置
[image]
make的结果
[image]

Groupsun, 2020-05-18:
> `Failed to import module fulladder_test: No module named 'fulladder_test'`

错误原因错在这里，你的python harness文件叫什么名字，是否确认是`fulladder_test.py`

sinceseto, 2020-05-18:
> > `Failed to import module fulladder_test: No module named 'fulladder_test'`
> 
> 错误原因错在这里，你的python harness文件叫什么名字，是否确认是`fulladder_test.py`

确实是这里错了，已解决，谢谢。

closed 2020-09-16 by Groupsun