Groupsun/CCPA-2020-LAB #21: 关于verilator --cc Decoder.v --exe Decoder_Sim.cpp --trace命令行出现的问题

Currt-Wei, 2020-05-19:
[image]
为什么会报procedural assignment to wire这个错误呢

Currt-Wei, 2020-05-19:
具体错误是:
%Error-PROCASSWIRE: Decoder.v:9: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'out'
                               : ... In instance Decoder
            out = 8'b00000000;
            ^~~

Groupsun, 2020-05-19:
这个似乎是Verilator的一个bug，某些版本会编译特定如上图verilog语句的时候会报错。这里我提供我的Verilator编译相同文件的详细信息以及Verilator版本供参考：

```
➜  verilator verilator --cc Decoder.v --exe Decoder_Sim.cpp --trace --debug
Starting Verilator 4.020 2019-10-06 rev v4.018-62-gdfcd412e
- V3File.cpp:219:        --check-times failed: different command line
- V3ParseImp.cpp:227: parseFile: Decoder
  Preprocessing Decoder.v
- V3PreShell.cpp:152:     Reading Decoder.v
- V3ParseImp.cpp:274: Lexing Decoder.v
- V3LinkCells.cpp:196:Link Module: MODULE 0x7f9ac0c081b0 <e3#> {c1} u4=0x7f9ac0c082c0  Decoder  L0
dot -Tpdf -o ~/a.pdf obj_dir/VDecoder_001_linkcells.dot
- V3LinkLevel.cpp:50: modSortByLevel()
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_002_cells.tree
- V3LinkDot.h:40:     linkDotPrimary:
- V3LinkJump.cpp:267: linkJump:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_007_link.tree
- V3Param.cpp:863:    param:
- V3LinkDot.h:45:     linkDotParamed:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_009_paramlink.tree
- V3Dead.cpp:427:     deadifyModules:
- V3Width.cpp:4211:   width:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_011_width.tree
- V3Width.cpp:4254:   widthCommit:
- V3Const.cpp:2643:   constifyAllLive:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_013_const.tree
- V3Undriven.cpp:458: undrivenAll:
- V3AssertPre.cpp:152:assertPreAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_014_assertpre.tree
- V3Assert.cpp:416:   assertAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_015_assert.tree
- V3LinkLevel.cpp:102:wrapTop:
- V3Const.cpp:2616:   constifyAllLint:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_017_const.tree
- V3Inst.cpp:598:     dearrayAll:
- V3LinkDot.h:50:     linkDotArrayed:
- V3Begin.cpp:290:    debeginAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_020_begin.tree
- V3Tristate.cpp:1378:tristateAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_021_tristate.tree
- V3Unknown.cpp:478:  unknownAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_022_unknown.tree
- V3Inline.cpp:620:   inlineAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_023_inline.tree
- V3LinkDot.h:50:     linkDotArrayed:
- V3Const.cpp:2653:   constifyAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_025_const.tree
- V3Dead.cpp:435:     deadifyDTypes:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_026_deadDtypes.tree
- V3Inst.cpp:590:     instAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_027_inst.tree
- V3Const.cpp:2653:   constifyAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_028_const.tree
- V3Scope.cpp:419:    scopeAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_029_scope.tree
- V3LinkDot.h:55:     linkDotScope:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_030_linkdot.tree
- V3Const.cpp:2653:   constifyAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_031_const.tree
- V3Dead.cpp:443:     deadifyDTypesScoped:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_032_deadDtypesScoped.tree
- V3Case.cpp:497:     caseAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_033_case.tree
- V3Task.cpp:1483:    taskAll:
dot -Tpdf -o ~/a.pdf obj_dir/VDecoder_034_task_call.dot
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_035_task.tree
- V3Name.cpp:147:     nameAll:
- V3Unroll.cpp:501:   unrollAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_037_unroll.tree
- V3Slice.cpp:244:    sliceAll:
[... 91 lines cut ...]
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_081_const_cpp.tree
- V3Subst.cpp:401:    substituteAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_082_subst.tree
- V3Const.cpp:2625:   constifyCpp:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_083_const_cpp.tree
- V3Dead.cpp:451:     deadifyAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_084_deadAll.tree
- V3Reloop.cpp:260:   reloopAll:
- V3Depth.cpp:177:    depthAll:
- V3Branch.cpp:138:   branchAll:
- V3Cast.cpp:184:     castAll:
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_087_cast.tree
- V3CCtors.cpp:148:   cctorsAll:
- V3EmitCInlines.cpp:94:emitcInlines:
- V3EmitCSyms.cpp:872:emitcSyms:
- V3EmitC.cpp:3243:   emitcTrace:
- V3EmitC.cpp:3229:   emitc:
- V3EmitXml.cpp:339:  emitxml:
- V3StatsReport.cpp:241:statsReport:
- V3EmitMk.cpp:249:   emitmk:
- V3Os.cpp:58:        export VERILATOR_ROOT=/usr/local/Cellar/verilator/4.020/share/verilator # Hardcoded at build time
- V3Os.cpp:58:        export SYSTEMC_ARCH=linux # From sysname 'darwin'
- V3Ast.cpp:1063:     Dumping obj_dir/VDecoder_990_final.tree
- Verilator.cpp:655:  Done, Exiting...
```

Currt-Wei, 2020-05-19:
@Groupsun 我去verilator的官网看了一下，别人之前发过这个问题，他说是“ The problem here is that there is a procedural assignment to a wire. The variable "out" should be declared as a reg or logic.” 好像是说Decoder——Sim_cpp文件里面输出out必须是一个reg或者logic，但是不知道要怎么改.... 而且我的verilator版本是最新的了

Groupsun, 2020-05-19:
如果搞不定的话可以先不管它，这个只是个例子，你用pyhcl写出来能够综合成功的放在verilator都是能编译成功的，这里给的例子主要是让你们知道harness该怎么写。

Groupsun, 2020-05-19:
或许你可以试试把`output [7:0] out`改成`output reg [7:0] out`

Currt-Wei, 2020-05-19:
@Groupsun 噢，改了这个之后就可以了，想想学长

Currt-Wei, 2020-05-19:
> 或许你可以试试把`output [7:0] out`改成`output reg [7:0] out`

改了这个之后就可以了，谢谢学长

closed 2020-09-16 by Groupsun