Groupsun/CCPA-2020-LAB #22: 环境配置？

CPJ31415, 2020-05-19:
[image]
[image]

我应该是配置好了，如第一张图。但是我关闭终端再打开就变成第二章图的样子了。然后再添加，又好了，关闭再打开又没了。

[image]
 
但是这个environment里面一直都有这个路径

CPJ31415, 2020-05-19:
@Groupsun

Groupsun, 2020-05-19:
你是使用了修改`/etc/environment`的方法是吗？这种方式似乎会有一点问题，之前也有同学问过我类似的问题，这种方法修改后似乎会被`/etc/profile`里面的内容覆盖掉。
要永久生效的话，最好使用修改`/etc/profile`或者`~/.bashrc`的方式。

CPJ31415, 2020-05-19:
路径的确出错了。重启后一直循环在输入密码的登陆界面。用Ctrl + Alt + F3 打开后，想通过vim /ect/proflie 修改路径。但是vim指令没找到，sudo的指令也没有找到。请问这个怎么解决？
@Groupsun

CPJ31415, 2020-05-19:
重启后一直循环在输入密码的登陆界面，用Ctrl + Alt + F3打开后，想通过vim / ect / proflie修改路径。但是vim指令没找到，sudo的指令也没有找到。请问这个怎么解决？ @Groupsun

Groupsun, 2020-05-19:
`vim`是一个终端用的文本编辑器，你需要安装：`sudo apt-get install vim`

CPJ31415, 2020-05-19:
但是sudo :command no found

Groupsun, 2020-05-19:
> 但是sudo :command no found

不是很明白你的意思，如果你已经在root用户下，那就不用`sudo`了。

CPJ31415, 2020-05-19:
就是, 因为路径改了。登不进去，我就用Ctrl + Alt + F3 打开。输入 vim 想改路径，提示no found。我也试了getit，也是no found。最后我想要sudo vim，也是command no found

Groupsun, 2020-05-19:
那你可能是这种情况，你看看这里：https://blog.csdn.net/qq_26442553/article/details/79183260

closed 2020-09-16 by Groupsun