Groupsun/CCPA-2020-LAB #26: 关于全加器级联时输出端位宽的问题

TEF-47, 2020-05-19:
在四位全加器的实现中，将四位输出用s=Output(U.w(4))，并在逻辑中定义
io.s[index] <<= FullAdders[index].io.s  ,
在生成.v文件时firrtl检查出错，提示 Expression _T_3（_T_7 _T_11 _T_15） is used as a SinkFlow but can only be used as a SourceFlow.
而使用 s=Output(Vec(4,U.w(1)))则不会有此错误，请问这是为何？

Groupsun, 2020-05-19:
取位操作只能作为右表达式使用，而不能作为左表达式。这是因为取位操作后在firrtl当中是作为常数来看待的，不能够修改它的值。
你还可以使用这种方法：

```python
io.s <<= CatBits(<由各个子全加器的输出口s组成的列表>)
```

Groupsun, 2020-05-19:
若你使用的是`Vec`，则此时取位操作得到的是其中每个输出口实体，这时候不会发生常量问题。

TEF-47, 2020-05-19:
io.s <<= CatBits([Adders[0].io.s,Adders[1].io.s,Adders[2].io.s,Adders[3].io.s])
是这样子用吗
但是会出错
AttributeError: 'list' object has no attribute 'mapToIR'

Groupsun, 2020-05-19:
如果你是这样写的话就改成：

```
CatBits(Adders[0].io.s, Adders[1].io.s, Adders[2].io.s, Adders[3].io.s)
```

你还要注意高低位的问题。

KaranaLJX, 2020-05-23:
使用vec还是报错
```python
class FullAdder(Module):
    io = IO(
        x=Input(U.w(4)),
        y=Input(U.w(4)),
        s=Output(Vec(5, U.w(1))),
        cout=Output(Vec(4, U.w(1)))
    )
    io.s.value[0] <<= io.x[0] ^ io.y[0]
```
```
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.s[1] <= VOID
```

Groupsun, 2020-05-23:
`value`是什么东西？并且输出口`s`都没有全部赋值。

KaranaLJX, 2020-05-23:
不加`value`python会报错，然后我用`dir(io.s)`列出他的成员,推测出`value`才是`Vec`的引用

KaranaLJX, 2020-05-23:
加了value之后就变成了firrtl报错
```
Exception in thread "main" firrtl.passes.PassExceptions: 
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.cout[3] <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.cout[0] <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.s[3] <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.s[0] <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.cout[2] <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.s[2] <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference _T_7 is not fully initialized.
   : _T_7[0] <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.cout[1] <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.s[4] <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module FullAdder]  Reference io is not fully initialized.
   : io.s[1] <= VOID
firrtl.passes.PassException: 10 errors detected!
```

Groupsun, 2020-05-23:
`s`是一个向量，你将不是向量的东西赋值过去肯定会报错。你得对`s`分开来赋值：

```python
io.s[0] <<= ???
io.s[1] <<= ???
...
```

然后你上面是没有将所有的输出口都接上去，输出口不能够悬空。通俗点说就是你的输出口没有全部赋值。

KaranaLJX, 2020-05-23:
```python
from pyhcl import *

class FullAdder(Module):
    io = IO(
        x=Input(U.w(4)),
        y=Input(U.w(4)),
        s=Output(Vec(5, U.w(1))),
        cout=Output(Vec(4, U.w(1)))
    )
    io.s[0] <<= io.x[0] ^ io.y[0]
    io.cout[0] << io.x[0] & io.y[0]
    for i in range(1, 4):
        io.s[i] <<= io.x[i] ^ io.y[i] ^ io.cout[i-1]
        io.cout[i] <<= io.x[i] & io.y[i] | io.x[i] & io.cout[i -
                                                             1] | io.y[i] & io.cout[i-1]
    io.s[4] <<= io.cout[3]
```
然后现在报错
```
Reference io is not fully initialized.
   : io.cout[0] <= VOID
```

closed 2020-05-23 by TEF-47