Groupsun/CCPA-2020-LAB #28: 生成verilog文件时遇到的问题

hzx718, 2020-05-20:
如图所示
[image]

hzx718, 2020-05-20:
完整代码如下，也不知道写得对不对：
from pyhcl import *
class FullAdder(Module):
    io = IO(
        a=Input(U.w(1)),
        b=Input(U.w(1)),
        cin=Input(U.w(1)),
        s=Output(U.w(1)),
        cout=Output(U.w(1))
    )
    # 填充完整输出端⼝s以及cout的逻辑，例如：
    io.s <<= io.a^io.b^io.cin
    io.cout <<= (io.a&io.b)|(io.a&io.cin)|(io.b&io.cin)
class FourAdder(Module):
    io=IO(
        a0=Input(U.w(1)),
        a1=Input(U.w(1)),
        a2=Input(U.w(1)),
        a3=Input(U.w(1)),
        b0=Input(U.w(1)),
        b1=Input(U.w(1)),
        b2=Input(U.w(1)),
        b3=Input(U.w(1)),
        s0=Output(U.w(1)),
        s1=Output(U.w(1)),
        s2=Output(U.w(1)),
        s3=Output(U.w(1)),
        cout=Output(U.w(1))
    )
    f0=FullAdder()
    f1=FullAdder()
    f2=FullAdder()
    f3=FullAdder()

    io.a0 <<= f0.io.a
    io.a1 <<= f1.io.a
    io.a2 <<= f2.io.a
    io.a3 <<= f3.io.a

    io.b0 <<= f0.io.b
    io.b1 <<= f1.io.b
    io.b2 <<= f2.io.b
    io.b3 <<= f3.io.b

    io.s0 <<= f0.io.s
    io.s1 <<= f1.io.s
    io.s2 <<= f2.io.s
    io.s3 <<= f3.io.s

    f0.io.cin <<= 0
    f1.io.cin <<= f0.io.cout
    f2.io.cin <<= f1.io.cout
    f3.io.cin <<= f2.io.cout

if __name__ == '__main__':
 Emitter.dumpVerilog(Emitter.dump(Emitter.emit(FourAdder()),"FourAdder.fir"))

Groupsun, 2020-05-20:
```python
f0.io.cin <<= 0
```
改为
```python
f0.io.cin <<= U(0)
```

hzx718, 2020-05-20:
好的，谢谢助教

closed 2020-09-16 by Groupsun