Groupsun/CCPA-2020-LAB #29: 执行make之后出现问题

Hollow-cat, 2020-05-20:
老师您好，我在测试chapter0中cocotb的例子时执行make命令出现以下问题，想知道如何解决。
[image]

Groupsun, 2020-05-20:
请提供更详细的执行后终端的信息，粘贴下来吧。

Hollow-cat, 2020-05-20:
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make results.xml
make[1]: Entering directory '/home/penghu/exp/2'
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make -C /home/penghu/.local/lib/python3.8/site-packages/cocotb/share/lib/utils SIM=verilator 
make[2]: Entering directory '/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/lib/utils'
make[2]: python: Command not found
make[2]: python: Command not found
make[2]: python: Command not found
make[2]: python: Command not found
make[2]: python: Command not found
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return -DVERILATOR -c -I/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/include -I  -o /home/penghu/exp/2/build/obj/x86_64/cocotb_utils.o cocotb_utils.c
gcc: error: /home/penghu/exp/2/build/obj/x86_64/cocotb_utils.o: No such file or directory
make[2]: *** [/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.rules:57: /home/penghu/exp/2/build/obj/x86_64/cocotb_utils.o] Error 1
make[2]: Leaving directory '/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/lib/utils'
make[1]: *** [/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/lib/Makefile:64: /home/penghu/exp/2/build/libs/x86_64/libcocotbutils.so] Error 2
make[1]: Leaving directory '/home/penghu/exp/2'
make: *** [/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.sim:80: sim] Error 2

Hollow-cat, 2020-05-20:
> 请提供更详细的执行后终端的信息，粘贴下来吧。

make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make results.xml
make[1]: Entering directory '/home/penghu/exp/2'
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make -C /home/penghu/.local/lib/python3.8/site-packages/cocotb/share/lib/utils SIM=verilator 
make[2]: Entering directory '/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/lib/utils'
make[2]: python: Command not found
make[2]: python: Command not found
make[2]: python: Command not found
make[2]: python: Command not found
make[2]: python: Command not found
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return -DVERILATOR -c -I/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/include -I  -o /home/penghu/exp/2/build/obj/x86_64/cocotb_utils.o cocotb_utils.c
gcc: error: /home/penghu/exp/2/build/obj/x86_64/cocotb_utils.o: No such file or directory
make[2]: *** [/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.rules:57: /home/penghu/exp/2/build/obj/x86_64/cocotb_utils.o] Error 1
make[2]: Leaving directory '/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/lib/utils'
make[1]: *** [/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/lib/Makefile:64: /home/penghu/exp/2/build/libs/x86_64/libcocotbutils.so] Error 2
make[1]: Leaving directory '/home/penghu/exp/2'
make: *** [/home/penghu/.local/lib/python3.8/site-packages/cocotb/share/makefiles/Makefile.sim:80: sim] Error 2

Groupsun, 2020-05-20:
终端python命令出问题了，你首先检查python的路径还不是出了什么问题，甚至，检查有没有配置python环境。

Hollow-cat, 2020-05-20:
好的谢谢老师，我去检查一下，您早点休息。

closed 2020-09-16 by Groupsun