Groupsun/CCPA-2020-LAB #30: 运行verilator --cc HalfAdder.v --exe HalfAdder_Sim.cpp --trace时遇到的问题

hzx718, 2020-05-20:
请问为什么会出现如图的错误
[image]
是漏了什么环境变量没有配置吗

hzx718, 2020-05-20:
HalfAdder_Sim.cpp代码如下:
`#include<verilated.h>
#include<berilated_vcd_c.h>
#include<VHalfAdder.h>
#include<iostream>
using namespace std;

VHalfAdder*top;
VerilatedVcdC* tfp;

vluint64_t main_time=0;
const vluint64_t sim_time=1024;
int main(int argc,char **argv)
{
    Verilated::commandArgs(argc,argv);
    Verilated::traceEverOn(true);
    top=new VHalfAdder;
    tfp= new VerilatedVcdC;
    top->trace(tfp,99)；
    tfp->open("VHalfAdder.vcd");

    int count1=0;
    int count2=0;
    while(!Verilated:gotFinish()&& main_time< sim_time)
    {
        top->reset=0;
        top->io_a=count1;
        top->io_b=count2;
        count1++;
        count2=count1-1;
        tfp->dump(main_time);
        main_time++;
    }
    
    tfp->close();
    delete top;
    delete tfp;
    exit(0);

}`

Groupsun, 2020-05-20:
`HalfAdder.v` -> `HalfAdder.fir.v`

hzx718, 2020-05-20:
试了一下，还是一样的错误

hzx718, 2020-05-21:
早上起来试了一下，发现是没有进入对应的文件夹所以显示找不到模块。但是运行时有关于#include的报错如下图，好像是没有找到berilated_vcd_c.h，最终生成的文件夹里面也没有找到vcd文件，请问是漏了什么环境变量没有设置还是因为什么原因呢
[image]
[image]

Groupsun, 2020-05-21:
头文件是`verilated_vcd_c.h`

hzx718, 2020-05-21:
我傻了....谢谢助教

closed 2020-09-16 by Groupsun