Groupsun/CCPA-2020-LAB #31: 生成文件夹中无法找到vcd文件

hzx718, 2020-05-21:
如图，按照教程运行完三条指令后找不到vcd文件，中途没有任何报错：
[image]
[image]

hzx718, 2020-05-21:
HalfAdder_sim代码如下：
#include<verilated.h>
#include<verilated_vcd_c.h>
#include<VHalfAdder.h>
#include<iostream>
using namespace std;

VHalfAdder*top;
VerilatedVcdC* tfp;

vluint64_t main_time=0;
const vluint64_t sim_time=1024;
int main(int argc,char **argv)
{
    Verilated::commandArgs(argc,argv);
    Verilated::traceEverOn(true);
    top=new VHalfAdder;
    tfp= new VerilatedVcdC;
    top->trace(tfp,99);
    tfp->open("VHalfAdder.vcd");

    int count1=0;
    int count2=0;
    while(!Verilated::gotFinish()&& main_time< sim_time)
    {
        top->reset=0;
        top->io_a=count1;
        top->io_b=count2;
        count1++;
        count2=count1-1;
        tfp->dump(main_time);
        main_time++;
    }
    
    tfp->close();
    delete top;
    delete tfp;
    exit(0);

}

Groupsun, 2020-05-21:
因为.vcd文件生成在了`.fir`文件夹下

hzx718, 2020-05-21:
好的，谢谢助教

closed 2020-09-16 by Groupsun