Groupsun/CCPA-2020-LAB #32: 运行实验报错

ZoeDong, 2020-05-22:
在运行实验的时候pycharm报错sh: 1: firrtl not found ，是在用户目录下执行的，也重启过虚拟机还是不行
[image]

环境变量感觉也没有问题
[image]

ZoeDong, 2020-05-22:
可以生成.fir文件 但是运行.py文件还是会报错

Groupsun, 2020-05-22:
报什么错误

ZoeDong, 2020-05-23:
和第一张图一样还是 sh: 1: firrtl not found

Groupsun, 2020-05-23:
那为什么可以生成`.fir`文件。同学能不能把你的问题说明的详细点。

ZoeDong, 2020-05-23:
不太清楚为什么会这样，我新建一个项目后新建一个python file，把实验内容复制进去运行，就可以生成.fir文件但是下方也会报错sh: 1: firrtl not found，不管是重启还是重新生成一个新项目都是既可以生成.fir文件也会报错
[image]

Groupsun, 2020-05-23:
你参考一下这个issue吧：https://github.com/Groupsun/CCPA-2020-LAB/issues/13

ZoeDong, 2020-05-23:
请问添加环境变量是在Run->Edit Configurations->Environment variables把PATH添加进去吗？如果是的话尝试过清除缓存和添加环境变量，但是都没有效果。但是又确实是能够生成.fir文件所以不太明白这个报错到底有没有影响，运行后界面是这样的。

[image]

closed 2020-09-16 by Groupsun