Groupsun/CCPA-2020-LAB #33: Verilator第二步make出现段错误

Onezity, 2020-05-23:
运行`make -j -C ./obj_dir -f VFourAdder.mk VFourAdder`
之前运行样例也会遇到段错误
[image]

Groupsun, 2020-05-23:
猜测你的C++文件有问题，提供你的C++代码。

Onezity, 2020-05-23:
https://paste.ubuntu.com/p/VnG5Qfh8KC/

Groupsun, 2020-05-23:
这看着似乎没什么问题，你试试看在make命令里加上`--debug`试试。

Onezity, 2020-05-23:
会显示找不到文件，但文件夹下确实有。
[image]
[image]

Groupsun, 2020-05-23:
你把终端所有信息粘贴下来给我看看

Onezity, 2020-05-23:
不知道为什么重启电脑后就能成功运行了，刚刚也没有安装什么。

Onezity, 2020-05-24:
助教您好，这个代码和上面完全一样，昨天重启后运行成功了，但现在重新生成后这句话会出现如下报错，重启也没有用。

`$ make -j -C ./obj_dir -f VFourAdder.mk VFourAdder --debug
GNU Make 4.2.1
为 x86_64-pc-linux-gnu 编译
Copyright (C) 1988-2016 Free Software Foundation, Inc.
许可证：GPLv3+：GNU 通用公共许可证第 3 版或更新版本<http://gnu.org/licenses/gpl.html>。
本软件是自由软件：您可以自由修改和重新发布它。
在法律允许的范围内没有其他保证。
正在读入 makefiles...
正在更新 makefile....
更新目标....
 文件“VFourAdder”不存在。
必须重新制作目标“VFourAdder”。
make: 进入目录“/home/spa/PycharmProjects/work2/.fir/obj_dir”
g++    1.o verilated.o verilated_vcd_c.o VFourAdder__ALL.a    -o VFourAdder -lm -lstdc++ 
collect2: fatal error: ld terminated with signal 11 [段错误], core dumped
compilation terminated.
make: *** [VFourAdder.mk:63：VFourAdder] 错误 1
make: 离开目录“/home/spa/PycharmProjects/work2/.fir/obj_dir”
`

closed 2020-09-16 by Groupsun