Groupsun/CCPA-2020-LAB #43: 仿真时输出后面为何一直是E

RedLotus-jpg, 2020-06-05:
[image]

移位寄存器这么定义的话，为什么波形图是下面这样的
[image]
当输入从1变为0时，理论上不应该是E C 8 0这样变化的吗

Groupsun, 2020-06-05:
我做了一个和你类似的测试，并没有这样的情况出现，代码：

```python
class Test(Module):
    io = IO(
        i=Input(U.w(1)),
        o=Output(U.w(4))
    )

    R0 = RegInit(U.w(1)(0))
    R1 = RegInit(U.w(1)(0))
    R2 = RegInit(U.w(1)(0))
    R3 = RegInit(U.w(1)(0))

    R3 <<= R2
    R2 <<= R1
    R1 <<= R0
    R0 <<= io.i

    io.o <<= CatBits(R3, R2, R1, R0)
```

仿真波形：

[image]

你提供一下你的完整代码以及寄存器R0～R3的仿真波形图可以帮助我发现哪里出了问题。

RedLotus-jpg, 2020-06-05:
`
class Register4(Module):
    io = IO(
        cin=Input(U.w(1)),
        cout=Output(U.w(4))
    ) 
    R0 = RegInit(U.w(1)(0))
    R1 = RegInit(U.w(1)(0))
    R2 = RegInit(U.w(1)(0))
    R3 = RegInit(U.w(1)(0))
    R3<<=R2
    R2<<=R1
    R1<<=R0
    R0<<=io.cin
    io.cout<<= CatBits(R3, R2, R1, R0)

if __name__ == '__main__':
    Emitter.dumpVerilog(Emitter.dump(Emitter.emit(Register4()),
                                        "Register4"))
`
[image]

RedLotus-jpg, 2020-06-05:
```
from pyhcl import *

class Register4(Module):
    io = IO(
        cin=Input(U.w(1)),
        cout=Output(U.w(4))
    )
    R0 = RegInit(U.w(1)(0))
    R1 = RegInit(U.w(1)(0))
    R2 = RegInit(U.w(1)(0))
    R3 = RegInit(U.w(1)(0))
    R3<<=R2
    R2<<=R1
    R1<<=R0
    R0<<=io.cin
    io.cout<<= CatBits(R3, R2, R1, R0)

if __name__ == '__main__':
    Emitter.dumpVerilog(Emitter.dump(Emitter.emit(Register4()),
                                        "Register4"))
```
[image]

Groupsun, 2020-06-06:
你把你的pyhcl源码，编译得到的v代码以及仿真代码上传上来让我看看吧，这个有点奇怪。

RedLotus-jpg, 2020-06-06:
[源代码.zip](https://github.com/Groupsun/CCPA-2020-LAB/files/4739463/default.zip)

RedLotus-jpg, 2020-06-06:
> 你把你的pyhcl源码，编译得到的v代码以及仿真代码上传上来让我看看吧，这个有点奇怪。

[源代码.zip](https://github.com/Groupsun/CCPA-2020-LAB/files/4739467/default.zip)

Groupsun, 2020-06-06:
你的`sin`在仿真文件当中输入定义的不好，溢出了`sin`的输入范围（`sin`只能取0和1）
你将第33行：

```C++
count1++;
```

改为

```C++
count1 = 1 - count1;
```

即可。

closed 2020-09-16 by Groupsun