Groupsun/CCPA-2020-LAB #45: 关于Catbis

Raydon993, 2020-06-06:
由提供的pdf，CatBits 接受的参数可以是⼀个参数的列表，或者是⼀个List 类型的变量，这其中存
储有需要被组合的信号。
`io.output <<= CatBits(r3, r2, r1, r0)。`
但是在仿真时，明显r3、r2、r1、r0的值发生了改变，但是output却在一些情况下没有任何变化，请问这是为什么呢?
[image]

```
class ShiftRegister(Module):
    io = IO(
        sin=Input(U.w(1)),
        output=Output(U.w(4))
    )
    # 实现逻辑
    r0 = RegInit(U.w(1)(0))
    r1 = RegInit(U.w(1)(0))
    r2 = RegInit(U.w(1)(0))
    r3 = RegInit(U.w(1)(0))

    r0 <<= io.sin
    r1 <<= r0
    r2 <<= r1
    r3 <<= r2
    io.output <<= CatBits(r3, r2, r1, r0)
```

Groupsun, 2020-06-06:
这个问题与问题：https://github.com/Groupsun/CCPA-2020-LAB/issues/43
一致，请参照这里；另外，我发现你的输入在同一个时钟周期内不稳定，我估计是这里出现了问题。

closed 2020-06-06 by Groupsun

Raydon993, 2020-06-06:
> 这个问题与问题：#43
> 一致，请参照这里；另外，我发现你的输入在同一个时钟周期内不稳定，我估计是这里出现了问题。

助教好，我懂了怎么回事了，io_sin在harness文件中不能做累加赋值。