Groupsun/CCPA-2020-LAB #47: 寄存器切片问题

KaranaLJX, 2020-06-07:
```python
from pyhcl import *

class fifo(Module):
    io = IO(
        en=Input(Bool),
        v=Input(U.w(32)),
        out_en=Input(Bool),
        out=Output(U.w(32))
    )
    io.out <<= U(0)
    cnt = Reg(U.w(2))
    r = Reg(U.w(96))
    with when(io.out_en & io.en):
        with when(cnt):
            io.out <<= (r[((cnt-U(1))*U(32)):32])
```
报错
```
cd /home/karana/study/CmposedExp ; env /bin/python3 /home/karana/.vscode-server/extensions/ms-python.python-2020.5.80290/pythonFiles/lib/python/debugpy/no_wheels/debugpy/launcher 55609 -- /home/karana/study/CmposedExp/exp4/fifo/work.py 
line 13:31 extraneous input '(' expecting {'circuit', 'module', 'extmodule', 'input', 'output', 'UInt', 'SInt', 'Fixed', 'Clock', 'Analog', 'flip', 'parameter', 'reset', ')', 'wire', 'reg', 'with', 'mem', 'cmem', 'smem', 'mport', 'inst', 'of', 'node', 'is', 'invalid', 'skip', 'depth', 'reader', 'writer', 'readwriter', 'when', 'else', 'infer', 'read', 'write', 'rdwr', 'old', 'new', 'undefined', 'mux(', 'validif(', 'stop', 'printf', 'mux', 'validif', 'add(', 'sub(', 'mul(', 'div(', 'rem(', 'lt(', 'leq(', 'gt(', 'geq(', 'eq(', 'neq(', 'pad(', 'asUInt(', 'asAsyncReset(', 'asSInt(', 'asClock(', 'asFixedPoint(', 'asInterval(', 'shl(', 'shr(', 'dshl(', 'dshr(', 'cvt(', 'neg(', 'not(', 'and(', 'or(', 'xor(', 'andr(', 'orr(', 'xorr(', 'cat(', 'bits(', 'head(', 'tail(', 'incp(', 'decp(', 'setp(', 'wrap(', 'clip(', 'squz(', UnsignedInt, SignedInt, HexLit, OctalLit, BinaryLit, Id}
line 13:35 extraneous input '=' expecting {'circuit', 'module', 'extmodule', 'input', 'output', 'UInt', 'SInt', 'Fixed', 'Clock', 'Analog', 'flip', 'parameter', 'reset', ')', 'wire', 'reg', 'with', 'mem', 'cmem', 'smem', 'mport', 'inst', 'of', 'node', 'is', 'invalid', 'skip', 'depth', 'reader', 'writer', 'readwriter', 'when', 'else', 'infer', 'read', 'write', 'rdwr', 'old', 'new', 'undefined', 'mux(', 'validif(', 'stop', 'printf', 'mux', 'validif', 'add(', 'sub(', 'mul(', 'div(', 'rem(', 'lt(', 'leq(', 'gt(', 'geq(', 'eq(', 'neq(', 'pad(', 'asUInt(', 'asAsyncReset(', 'asSInt(', 'asClock(', 'asFixedPoint(', 'asInterval(', 'shl(', 'shr(', 'dshl(', 'dshr(', 'cvt(', 'neg(', 'not(', 'and(', 'or(', 'xor(', 'andr(', 'orr(', 'xorr(', 'cat(', 'bits(', 'head(', 'tail(', 'incp(', 'decp(', 'setp(', 'wrap(', 'clip(', 'squz(', UnsignedInt, SignedInt, HexLit, OctalLit, BinaryLit, Id}
line 13:39 extraneous input '(' expecting {'circuit', 'module', 'extmodule', 'input', 'output', 'UInt', 'SInt', 'Fixed', 'Clock', 'Analog', 'flip', 'parameter', 'reset', ')', 'wire', 'reg', 'with', 'mem', 'cmem', 'smem', 'mport', 'inst', 'of', 'node', 'is', 'invalid', 'skip', 'depth', 'reader', 'writer', 'readwriter', 'when', 'else', 'infer', 'read', 'write', 'rdwr', 'old', 'new', 'undefined', 'mux(', 'validif(', 'stop', 'printf', 'mux', 'validif', 'add(', 'sub(', 'mul(', 'div(', 'rem(', 'lt(', 'leq(', 'gt(', 'geq(', 'eq(', 'neq(', 'pad(', 'asUInt(', 'asAsyncReset(', 'asSInt(', 'asClock(', 'asFixedPoint(', 'asInterval(', 'shl(', 'shr(', 'dshl(', 'dshr(', 'cvt(', 'neg(', 'not(', 'and(', 'or(', 'xor(', 'andr(', 'orr(', 'xorr(', 'cat(', 'bits(', 'head(', 'tail(', 'incp(', 'decp(', 'setp(', 'wrap(', 'clip(', 'squz(', UnsignedInt, SignedInt, HexLit, OctalLit, BinaryLit, Id}
line 13:43 extraneous input '=' expecting {'circuit', 'module', 'extmodule', 'input', 'output', 'UInt', 'SInt', 'Fixed', 'Clock', 'Analog', 'flip', 'parameter', 'reset', ')', 'wire', 'reg', 'with', 'mem', 'cmem', 'smem', 'mport', 'inst', 'of', 'node', 'is', 'invalid', 'skip', 'depth', 'reader', 'writer', 'readwriter', 'when', 'else', 'infer', 'read', 'write', 'rdwr', 'old', 'new', 'undefined', 'mux(', 'validif(', 'stop', 'printf', 'mux', 'validif', 'add(', 'sub(', 'mul(', 'div(', 'rem(', 'lt(', 'leq(', 'gt(', 'geq(', 'eq(', 'neq(', 'pad(', 'asUInt(', 'asAsyncReset(', 'asSInt(', 'asClock(', 'asFixedPoint(', 'asInterval(', 'shl(', 'shr(', 'dshl(', 'dshr(', 'cvt(', 'neg(', 'not(', 'and(', 'or(', 'xor(', 'andr(', 'orr(', 'xorr(', 'cat(', 'bits(', 'head(', 'tail(', 'incp(', 'decp(', 'setp(', 'wrap(', 'clip(', 'squz(', UnsignedInt, SignedInt, HexLit, OctalLit, BinaryLit, Id}
line 13:47 extraneous input '(' expecting {'circuit', 'module', 'extmodule', 'input', 'output', 'UInt', 'SInt', 'Fixed', 'Clock', 'Analog', 'flip', 'parameter', 'reset', ')', 'wire', 'reg', 'with', 'mem', 'cmem', 'smem', 'mport', 'inst', 'of', 'node', 'is', 'invalid', 'skip', 'depth', 'reader', 'writer', 'readwriter', 'when', 'else', 'infer', 'read', 'write', 'rdwr', 'old', 'new', 'undefined', 'mux(', 'validif(', 'stop', 'printf', 'mux', 'validif', 'add(', 'sub(', 'mul(', 'div(', 'rem(', 'lt(', 'leq(', 'gt(', 'geq(', 'eq(', 'neq(', 'pad(', 'asUInt(', 'asAsyncReset(', 'asSInt(', 'asClock(', 'asFixedPoint(', 'asInterval(', 'shl(', 'shr(', 'dshl(', 'dshr(', 'cvt(', 'neg(', 'not(', 'and(', 'or(', 'xor(', 'andr(', 'orr(', 'xorr(', 'cat(', 'bits(', 'head(', 'tail(', 'incp(', 'decp(', 'setp(', 'wrap(', 'clip(', 'squz(', UnsignedInt, SignedInt, HexLit, OctalLit, BinaryLit, Id}
line 13:51 extraneous input '=' expecting {'circuit', 'module', 'extmodule', 'input', 'output', 'UInt', 'SInt', 'Fixed', 'Clock', 'Analog', 'flip', 'parameter', 'reset', ')', 'wire', 'reg', 'with', 'mem', 'cmem', 'smem', 'mport', 'inst', 'of', 'node', 'is', 'invalid', 'skip', 'depth', 'reader', 'writer', 'readwriter', 'when', 'else', 'infer', 'read', 'write', 'rdwr', 'old', 'new', 'undefined', 'mux(', 'validif(', 'stop', 'printf', 'mux', 'validif', 'add(', 'sub(', 'mul(', 'div(', 'rem(', 'lt(', 'leq(', 'gt(', 'geq(', 'eq(', 'neq(', 'pad(', 'asUInt(', 'asAsyncReset(', 'asSInt(', 'asClock(', 'asFixedPoint(', 'asInterval(', 'shl(', 'shr(', 'dshl(', 'dshr(', 'cvt(', 'neg(', 'not(', 'and(', 'or(', 'xor(', 'andr(', 'orr(', 'xorr(', 'cat(', 'bits(', 'head(', 'tail(', 'incp(', 'decp(', 'setp(', 'wrap(', 'clip(', 'squz(', UnsignedInt, SignedInt, HexLit, OctalLit, BinaryLit, Id}
line 13:59 extraneous input ''pyhcl.dsl.cdatatype.U2'' expecting {'circuit', 'module', 'extmodule', 'input', 'output', 'UInt', 'SInt', 'Fixed', 'Clock', 'Analog', 'flip', 'parameter', 'reset', ')', 'wire', 'reg', 'with', 'mem', 'cmem', 'smem', 'mport', 'inst', 'of', 'node', 'is', 'invalid', 'skip', 'depth', 'reader', 'writer', 'readwriter', 'when', 'else', 'infer', 'read', 'write', 'rdwr', 'old', 'new', 'undefined', 'mux(', 'validif(', 'stop', 'printf', 'mux', 'validif', 'add(', 'sub(', 'mul(', 'div(', 'rem(', 'lt(', 'leq(', 'gt(', 'geq(', 'eq(', 'neq(', 'pad(', 'asUInt(', 'asAsyncReset(', 'asSInt(', 'asClock(', 'asFixedPoint(', 'asInterval(', 'shl(', 'shr(', 'dshl(', 'dshr(', 'cvt(', 'neg(', 'not(', 'and(', 'or(', 'xor(', 'andr(', 'orr(', 'xorr(', 'cat(', 'bits(', 'head(', 'tail(', 'incp(', 'decp(', 'setp(', 'wrap(', 'clip(', 'squz(', UnsignedInt, SignedInt, HexLit, OctalLit, BinaryLit, Id}
line 13:90 no viable alternative at input 'rhs='
Exception in thread "main" firrtl.SyntaxErrorsException: 8 syntax error(s) detected
```

Groupsun, 2020-06-07:
寄存器切片里面的index不可以使用pyhcl类型，只能是python整型。

KaranaLJX, 2020-06-07:
那如果参数必须通过pyhcl的类型计算，就不能用切片了，只能通过右移和与实现？
切片的参数也很迷？a>b s[a:b]是从前a往后取到b，还是从a往后取b位

Groupsun, 2020-06-08:
s[a:b]指切片从a取到b：如`r = 0x1234`，则`r[3:0] = 0x4`；

closed 2020-09-16 by Groupsun