Groupsun/CCPA-2020-LAB #49: 关于激活函数的问题

sinceseto, 2020-06-15:
> 激活函数有多种实现方式，这里要求实现ReLU的激活函数，它的定义如下：
> $$
> relu(x)=
> \begin{cases}
> 0, & x \leq 0 \\
> x, & x > 0
> \end{cases}
> $$
> 在讨论如何实现MAC之前，首先来讨论python在函数化编程中会使用到的高阶函数。
输入端口都是无符号数，激活函数的输入是乘加的运算结果，应该都是非负的，那这个激活函数没意义吧？需要实现过程是什么意思？

Groupsun, 2020-06-15:
实际上输入定义为无符号数，如果在后续的代码中将它作为有符号数看待，那么就有可能是负数。
比如：

```python
class Test(Module):
    io = IO(
        vin=Input(U.w(4)),
        output=Output(U.w(4))
    )

    with when(io.vin.to_sint() < S(0)):
        io.output <<= U(1)
    with otherwise():
        io.output <<= U(0)
```

我将输入作为有符号数来看待，`to_sint()`是类型转换。那么当我输入-1的时候，输入的值是0xF。如果我作为无符号数来看待，那么`output`必然输出为0。但是如果我像上述代码作为有符号数来看待，那么就会输出为1。

总的来说，类型的有无符号只是一种形式上的约束（比如有符号和无符号之间不能直接运算），实际上，真正实在的东西是这个位串，但位串你选择用什么方式去解释在不同的情况都不一样了。

另外，`S()`这个api有一个小bug需要修复一下，可以参照最新的commit：https://github.com/scutdig/PyChip-py-hcl

更新你电脑的PyHCL库，直接pull或者clone最新版本即可。

Groupsun, 2020-06-15:
https://github.com/scutdig/PyChip-py-hcl/commit/74311bbe80a136c1cbebedb87382e427f23194f3

closed 2020-09-16 by Groupsun